smoochiee/Noisy-boy-esp32-Bluetooth-jammer #3: This isnt even an issue

Lgndwilloske, 2024-09-29:
I have to say, there must of been some time and effort put into this. The code is awesome and it works brilliantly.

Thanks - Willy

smoochiee, 2024-09-30:
> I have to say, there must of been some time and effort put into this. The code is awesome and it works brilliantly.
> 
> Thanks - Willy

thanks

Marakesh85, 2024-10-24:
> I have to say, there must of been some time and effort put into this. The code is awesome and it works brilliantly.
> 
> Thanks - Willy

I wonder how you tested it. Ive tested it with my Anker bluetooth speaker and it has no effect. 
Any special configurations youve done?
I cant get it to work.

Tested with wroom32u
2x Rnf24l01 pn+lna

Lgndwilloske, 2024-10-24:
aye, I had a August headset that i tested, same hardware as you maybe reflash?

Marakesh85, 2024-10-25:
> aye, I had a August headset that i tested, same hardware as you maybe reflash?

Ive Reflashed it serveral times.^^ i dont know why it doesnt work. Im going to test it with some other devices

Lgndwilloske, 2024-10-25:
your clicking the boot button?

Marakesh85, 2024-10-25:
> your clicking the boot button?

Jepp.

smoochiee, 2024-10-26:
> > your clicking the boot button?
> 
> Jepp.

you need to configure it first...follow step 3

Marakesh85, 2024-10-27:
I had it configured :/ im going to wire it again directly to the esp with jumper wires instead of breadboard connections. I also have a soldered one that is also not working. The thing is that i have no Option to test the nrf modules if they really work or if they are damaged.-_-

smoochiee, 2024-10-27:
> I had it configured :/ im going to wire it again directly to the esp with jumper wires instead of breadboard connections. I also have a soldered one that is also not working. The thing is that i have no Option to test the nrf modules if they really work or if they are damaged.-_- 

i hate breadboard.. sometimes they are not working😂

dgorbatko, 2024-11-07:
Hi, quick question, i have M5stickc plus 2 based on esp32. But it has limited connectors. So i can physicaly connect only one NRF24 (I connected and it works). So to get stronger jamming effect, like you did with dual nrf24 (but you did it on two separated channels), so if i connect two nrf24 in parallel to my esp32 m5stickc pluse 2, will it work? And if it works will the jamming effect be stornger?

smoochiee, 2024-11-08:
the jamming effect will be the same...same channel...does not make sense...if you want to increase range buy amplifier...

Kaanseniyesin, 2024-11-09:
i can see it transmitting and then it stops transmitting i am using esp32 30 pin development with nrf24l01

closed 2024-11-16 by smoochiee