smoochiee/Noisy-boy-esp32-Bluetooth-jammer #5: Boot Button

Lgndwilloske, 2024-10-11:
How do i change theboot button?? and what can i change it to??

smoochiee, 2024-10-12:
> How do i change theboot button?? and what can i change it to??

add button on GPIO4 and flash the new firmware https://smoochiee.github.io/Noisy-boy-esp32-Bluetooth-jammer/flash

Lgndwilloske, 2024-10-12:
so, do i solder 2 wires to gpio4?

smoochiee, 2024-10-13:
> so, do i solder 4 wires to gpio4? the button i have has 4 solder points
you can search google on how to do it...

Lgndwilloske, 2024-10-13:
There is nothing online that helps.

Lgndwilloske, 2024-10-13:
The button i have has 4 pins, I have soldered one to gpio 4 what about the second one?

Lgndwilloske, 2024-10-13:
It also doesent jam anything for some reason, I have sorted out the boot button btw it'd gpio 4 and ground

smoochiee, 2024-10-13:
> It also doesent jam anything for some reason, I have sorted out the boot button btw it'd gpio 4 and ground

again..you need to configure it on webrowser .follow instructions..read

closed 2024-10-18 by smoochiee