smoochiee/Noisy-boy-esp32-Bluetooth-jammer #9: Not really an issue but a question

8JustMe8, 2024-12-04:
Hello, first of all great work and thank you. Is there a way to program the spam to be activated from the web interface instead of the boot button ? Sorry if this was asked before, I really tried finding the answer myself before posting but to no avail. Thank you for your time.

smoochiee, 2024-12-07:
> Hello, first of all great work and thank you. Is there a way to program the spam to be activated from the web interface instead of the boot button ? Sorry if this was asked before, I really tried finding the answer myself before posting but to no avail. Thank you for your time.

implemented it on first release but i dont like it...

Casao16, 2024-12-09:
Boa Tarde, tentei fazer essa ligação e usar o comando no [WEBFLASHER] e no  https://esp.huhn.me/ e não consegui fazer o meu ESP32 WROOM 32 ( 38 PINOS ) funcionar, teria algum outro video demostrando no esp de 38 pinos ? Grato.

smoochiee, 2024-12-11:
> Boa Tarde, tentei fazer essa ligação e usar o comando no [WEBFLASHER] e no  https://esp.huhn.me/ e não consegui fazer o meu ESP32 WROOM 32 ( 38 PINOS ) funcionar, teria algum outro video demostrando no esp de 38 pinos ? Grato.

if you compare pins of 38 and 30 if same it will work...take note of the buttons

closed 2024-12-31 by smoochiee