smoochiee/Noisy-boy-esp32-Bluetooth-jammer #11: No bluetooth jamming.

javi-el-largo, 2025-01-08:
Hi, I think it’s an amazing project. I flashed the noisyboy ‘boot_button’ firmware on my ESP-32. The blue LED flashes, and I can access the web server on 192.168.4.1. I followed your wiring for two NRF24L01 modules, I power then with 2 3.3v regulators. I’ve checked all the connections with a multimeter and the’re fine, no short-circuits. I’ve tried several RF settings with no success. I’m not able to jam my bluetooth speaker that is 1 meter away. Can you help
me please? What settings do I need to use? My RF modules have PA/LNA. Thank you very much.

z3r0l1nk, 2025-01-08:
I also did this project, in the first try, I mixed up the wires and nothing was working, after a closer inspection and fixing my bad wire mixup. Everything worked like a charm.
Note to dev, could you write an app for the flipper zero on the Momentum Firmware? That would be the awesome.

javi-el-largo, 2025-01-08:
I’ve checked the wiring four times, there is no shortcut circuit, all the pins on the RF modules are connected to their respective ESP32 pins, the RF modules are receiving 3.29 volts… I think it must be something about RF settings.

closed 2025-01-14 by smoochiee