smoochiee/Noisy-boy-esp32-Bluetooth-jammer #12: Wrong VSPI 'CS' pin

javi-el-largo, 2025-01-13:
According to the official ESP-WROOM-32 pinout, the right VSPI CS pin is D5, NOT RX2 as you described.  
[image]

smoochiee, 2025-01-13:
> According to the official ESP-WROOM-32 pinout, the right VSPI CS pin is D5, NOT D21 (I2C SDA) as you described.  
> [image]
> 

idk to your setup..as anyone have made it work and i been using this...user problem...

smoochiee, 2025-01-13:
> According to the official ESP-WROOM-32 pinout, the right VSPI CS pin is D5, NOT D21 (I2C SDA) as you described.  
> [image]
> 

you follow the pin out on the readme😞theirs no problem with code and pins...its your build problem

javi-el-largo, 2025-01-13:
Followed your wiring, checked everything many times, tried with D5, with D21, tried with 3 different ESO-WROOM-32 boards, only work as WI-FI jammer, no bluetooh jamming. I give up. I’ll try another project like Bruce Predatory.