smoochiee/Noisy-boy-esp32-Bluetooth-jammer #25: How to improve range?

JesusMF23, 2025-08-09:
I am having trouble with range. Only with like 3-4 meters and couple of walls in between it stops working in bluetooth jamming. Anyone knows how to improve range either adding new modules, different modules or better antenna? I need it to work at least to 20-30 meters with couple of walls in between so I can make my neighbor stop the music without needing to be on its door.

JesusMF23, 2025-08-09:
I am currently at max. I imagine is more a hardware thing. Do you have any tips for using better antennas or better nrf modules? Or maybe using more than 2 modules?

smoochiee, 2025-08-09:
buy directional antenna...search google you will see what it is... ebyte e01-2g4m27d..make sure have stable psu...

djmuerteitm, 2025-08-15:
> I am currently at max. I imagine is more a hardware thing. Do you have any tips for using better antennas or better nrf modules? Or maybe using more than 2 modules?

I has build my own jammer using two 2g4m27d (genuine Ebyte) modules, with two nrf24l01 adapter.
This simplified setup works using 5vdc line for all stuff (esp and modules), and ensures feeds it with enough power.

You only need the adapter if you plan use your jammer with BOTH standard nrf24 or Ebyte modules. If you only want to use Ebyte, can be powered directly at 5v without adapter. On my setup, i use adapters for fast replacement if any module fails, like a hot-plug.

Esp board and adapters includes a onboard AMS1117 regulator. With 5V input, it provides constant 1A current at 3.3V.

2G4m27d modules, demands max 490mA at 3.3V and 390mA at 5V (manufacturer specs). 

If you use a external battery (typical power bank) as power source, keep attention on its power delivery specs. If power bank can provide at least 2A (real) output and if you use a proper USB cable, it must be enough to feed a 1 esp mcu + 2 2g4 setup with oled display for sure.

[image]

[image]

[image]

My setup as described