smoochiee/Noisy-boy-esp32-Bluetooth-jammer #26: Range is only 3m

JesusMF23, 2025-08-14:
With the following setup and even adding bigger antennas, 3m and is no longer working. I measured voltage on nrf modules and between VIN and GND I have 3.2V so it looks good.

I don't know if something is wrong, I have doubts regarding the capacitors you used on an old version or why the range is so low. 
[image]
My settings are the following:
[image]

cobra20g, 2025-08-26:
You need better RF modules, and your wires are too long. That's what he's telling you.> With the following setup and even adding bigger antennas, 3m and is no longer working. I measured voltage on nrf modules and between VIN and GND I have 3.2V so it looks good.
> 
> I don't know if something is wrong, I have doubts regarding the capacitors you used on an old version or why the range is so low. 
> [image]
> 
> My settings are the following:
> [image]
> 
> my scheme of connections is:
> 
> [image]

JesusMF23, 2025-11-03:
I have installed the ebyte with caps and still not much range. this is the ebyte I bought: https://es.aliexpress.com/item/1005007755505718.html 

any other suggestion, is like 7-8m and is not working. could it really be the wiring?